lwfinger/rtl8723au_bt #20: kernel 3.18 compile error

sunny0, 2015-01-06:
Hi, I was just trying to get the bluetooth working with my annoying rtl8723be hardware, but it won't compile because hci_recv_fragment was removed.

e.g.
https://github.com/lwfinger/rtl8723au_bt/issues/16

Of course it's not meant for my hardware (seems like nothing is), but I thought i'd let you know.

closed 2015-01-08 by sunny0

freechelmi, 2015-01-08:
Not sure what you meant ? the rtl8723be is working great here with the 'new' branch.

You can give a try to my dkms package, I did not tested it on 3.18 but it should work : 

http://support.ekimia.fr/drivers/Bluetooth/rtl8x/

lwfinger, 2015-01-08:
sunny0: Just because the repo is named rtl8723au_bt does not mean that it is not for your rtl8723be BT hardware. There is code on there for the BT parts of RTL8723AU, RTL8723AE, RTL8723BU, RTL8723BE, RTL8821AE, RTL8812AU, RTL8812AE, and probably others.

As I told you in a separate thread, I have not updated the master branch for newer kernels, mainly because it does not work. Use the 'new' branch!!

sunny0, 2015-01-09:
that's why i closed it a few days ago...