lwfinger/rtl8723au_bt #21: Troy branch compil failed

Zophar78, 2015-02-01:
I got the following error during compile (On 3.13 and 3.17 kernel):

*make -C /lib/modules/3.13.0-44-generic/build M=/home/pierre/git/rtl8723au_bt modules
make[1]: Entering directory `/usr/src/linux-headers-3.13.0-44-generic'
  CC [M]  /home/pierre/git/rtl8723au_bt/btusb.o
/home/pierre/git/rtl8723au_bt/btusb.c:215:12: error: static declaration of ‘hci_recv_fragment’ follows non-static declaration
 static int hci_recv_fragment(struct hci_dev *hdev, int type, void *data, int count)
            ^
In file included from /home/pierre/git/rtl8723au_bt/btusb.h:39:0,
                 from /home/pierre/git/rtl8723au_bt/btusb.c:34:
include/net/bluetooth/hci_core.h:769:5: note: previous declaration of ‘hci_recv_fragment’ was here
 int hci_recv_fragment(struct hci_dev *hdev, int type, void *data, int count);*

lwfinger, 2015-02-01:
I fixed this function. Do a 'git pull' and it should build without error.

lwfinger, 2015-02-11:
FYI, there is a new 'kernel' branch that has code that will be in kernel 3.21. It handles the Realtek devices just like any other kernel device. Thus far, no changes for older kernels have been included. As problems are reported, I will fix them.

rzr, 2015-05-04:
I am unsure kernel.org is shiping BT features at least for 0bda:b728 ? 
How can we help to get it supported  on lenovo g40-30 ?

I am tracking my investigations at :
http://rzr.online.fr/q/ideapad

lwfinger, 2015-05-04:
Clone the git repo at http://github.com/lwfinger/rtl8723au_bt.git, checkout the "kernel" branch, and build and install it. The 0bda:b728 device is there and I have shown that driver to work.

rzr, 2015-05-04:
I buit this repo  based on your changes : 

http://git.kernel.org/cgit/linux/kernel/git/bluetooth/bluetooth-next.git/commit/?id=a2698a9bf9b0cfcf7e556946e04c4456a13a6f5a

it worked ... thanks again for support