lwfinger/rtl8723au_bt #24: Bluetooth stops working after suspend

ohgodwhyyy, 2015-04-07:
Hello!

I installed Ubuntu 15.04 and with it the 4.0 RC6 Mainline Kernel. I compiled rtlwifi-new and rtl8723au_bt from the troy branch. I got bluetooth to work but every time after suspending I have to install my bluetooth mouse again. From the taskbar I can see that the mouse is trying to connect with my computer (connection lasts for less than a second every time I move the mouse).

I also tried the kernel branch of rtl8723au_bt, but the same thing happens.
Any recommendations on how to get bluetooth working like it did with the previous Ubuntu version?

ohgodwhyyy, 2015-04-07:
Please close, I was dumb not to click the "Trust device" button, after trusting everything works fine.

closed 2015-04-07 by lwfinger

ohgodwhyyy, 2015-04-08:
Okay, it seems that there is some kind of a problem after all. I left my laptop in suspend mode overnight and woke it up in the morning, my mouse wasn't working. Rebooting fixed it (no need to remove an readd the mouse via bluetooth settinsg), but if I suspend and immediatley resume, my mouse still works without any hassle.
So it seems to be a time-based issue.

ohgodwhyyy, 2015-04-08:
Added the script that's shown in this thread: http://ubuntuforums.org/showthread.php?t=1387211 and bluetooth worked after ~2 hours of suspend.

lwfinger, 2015-04-08:
From that script, it appears to be an rfkill problem, not one with the BT driver. Glad you found a fix.