lwfinger/rtl8723au_bt #25: not support for RTL8723BU bluetooth driver on kernel version 3.8.13

zz5414, 2015-05-11:
Hello

I successed porting rtl8723bu bluetooth driver on Beaglebone black with kernel version 3.13.6.

And this time I used kernel branch.

But can't compile bluetooth driver on kernel version 3.8.13 with kernel branch.

Without kernel version up, Are there any method to porting bluetooth driver on kernel version 3.8.13?

lwfinger, 2015-05-11:
Whenever I create a kernel branch, that means it is for the current kernel. Kernel 3.8 is hardly the current one.

I have no interest in backporting this driver to kernels that old. Cannot Beaglebone handle a much newer kernel?

zz5414, 2015-05-11:
Thanks.

I'll try to upgrade beaglebone kernel version.