lwfinger/rtl8723au_bt #26: rtl8723be (0bda:b001) bluetooth doesnt work on linux-4.0.4

bm16ton, 2015-05-30:
I have tried the kernel and the new branchs. dmesg shows a complaint that the firmware is for 8723 but card is 0x4ce1. Both branch's work on 3.19.x for me, on the 3.19.x branchs hciconfig -a hci0 show the LMP subversion as 0x4ce1 but there is no complaint about this in dmesg. I brought the bluetooth stack from the 3.19.x and replaced it with 4.0.4's, this came the closest to working i actually get a mac address and the bottom info from the hciconfig -a hci0 command. But even tho it tries without error hcitool scan plus gui tools are unable to find anything. I have made sure I have the correct firmwares in correct folders /lib/firmware /lib/firmware/rtl_bt  I even put the firmware in both places for each driver respectivly. Thank You;
PS originaly i could never get these drivers to work unless I also used lwfingers wifi driver. What seemed to happen is bluetooth would try to use non existent antenna (my card has only one antenna port for both wifi bluetooth, not sure if all 8723be cards are like this) and have tried both stock and lwfingers wifi driver

lwfinger, 2015-05-30:
We are currently working on problems with the single-antenna RTL8723BE models. A work-around was recently applied to the 4.1 kernel source. I have backported this to all branches of rtlwifi_new. The patch will eventually be backported to 4.0, but that will take a while.

Be careful in interpreting the LMP version as it changes when the firmware is loaded. As long as the driver is loaded without a complaint, then it is OK.

bm16ton, 2015-05-30:
Thank You! after removing the rtlwifi.ko's rebooting bluetooth works fine and hciconfig shows a previously unseen lmp subversion starting with 9 (sorry caint remember the rest) I see the rtlwifi source I was using in most of my 3.19.x kernels is slightly diff so will try this along with useing 3.19.3's bluetooth stack. and will also look into the patch for 4.1 see if I caint jerry rig it in. Thank you again!

bm16ton, 2015-05-30:
ps sorry for dumb question what exactly is the LMP subversion? google seems to alude me on this

bm16ton, 2015-05-30:
after rebooting from the kernel with the deleted rtlwifi.ko with working bluetooth into another 4.0.4 whos bluetooth never worked it now shows LMP subversion 0x9f73 and works!! Wthout any changes so im lost unless something in firmware survivied the reboot. Ill reboot, test again

bm16ton, 2015-05-30:
Works no more LMP subversion 0x4ce1 its now 0x9f73. Wifi and bluetooth work fine side by side. No changes at all to this kernel. Only difference is I booted a different 4.0.4 without rtlwifi support which fixed bluetooth. Must be coincidental but its working thats all that matters. Thank you for your help. If i figure it out ill post fix for others. Working 4.0.4 useing rtlwifi_new-master and rtl8723au_bt-kernel

colbyf, 2015-06-06:
Hi,
This git has been a life saver for me through my linux experience in the last 6 weeks. Although I have now hit a problem. Made the switch to Arch and I can't get Bluetooth workings, seems to be blind. I know I am using 4.04 kernel. If anyone can help it would be appreciated. I have gisted all I think that will help disgnosis, if you need more just ask. I think this may be the relevant thread.

https://gist.github.com/colbyf/84d3084fbd44ac7fb38e

lwfinger, 2015-06-06:
I do not use ARCH, but I would suspect something is the user code. See the error messages that say

Jun 06 11:02:37 archvoid bluetoothd[443]: Not enough free handles to regist...e
Jun 06 11:02:37 archvoid bluetoothd[443]: Not enough free handles to regist...e

These indicate a problem.

From other postings on the net, it appears that other ARCH users are having problems, particularly with bluez 5.30. I'm currently using 5.23, which is working with all kernels.

drzraf, 2015-07-31:
10ec:b723 (RTL8723BE), today's git of rtlwifi_new + troy's branch for bluetooth on Debian + 4.0.0-2-amd64 : I can get the bluetooth to function well (basic gnokii usage tested successfully)