lwfinger/rtl8723au_bt #30: Clarification on purpose of the different branches

psanford, 2015-06-05:
It was not clear to me which branch I should use for RTL8723BU with a 3.19 kernel. 

Testing I found the 'new' branch worked, but it would be nice if it were documented in the readme which branch should be used when. I'd be happy to open a pull request if someone can provide an outline of the purpose of different branches.

lwfinger, 2015-06-05:
Branch "master" has the original driver that I got from Realtek.
Branch "kernel" has the Daniel Drake patches to the kernel's btusb.
Branch "troy" has another driver that was provided by Realtek.
Branch "new" is yet another Realtek version.

There are so many because most of them do not work. Use the one that does.

psanford, 2015-06-05:
Thanks for the explanation.

closed 2015-06-05 by psanford