lwfinger/rtl8723au_bt #32: Driver don't seem to work on kernel 3.19.8-100.fc20.x86_64

mehanzhi, 2015-06-15:
Hello,

I'll just give you output, it's not that long and it has all I did. 

[root@localhost rtl8723au_bt-kernel]# make
make -C /lib/modules/3.19.8-100.fc20.x86_64/build M=/home/k_georgiev/work/rtl8723au_bt-kernel modules
make[1]: Entering directory `/usr/src/kernels/3.19.8-100.fc20.x86_64'
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: Leaving directory `/usr/src/kernels/3.19.8-100.fc20.x86_64'
[root@localhost rtl8723au_bt-kernel]# make install
depmod -a /lib/modules/3.19.8-100.fc20.x86_64
installed revised btusb
[root@localhost rtl8723au_bt-kernel]# modprobe -rv btusb
rmmod btusb
[root@localhost rtl8723au_bt-kernel]# modprobe -v btusb 
insmod /lib/modules/3.19.8-100.fc20.x86_64/kernel/drivers/bluetooth/btusb.ko 
[root@localhost rtl8723au_bt-kernel]# lsmod |grep -i btusb.
btusb                  45899  0 
[root@localhost rtl8723au_bt-kernel]# rfkill list
0: ideapad_wlan: Wireless LAN
	Soft blocked: no
	Hard blocked: no
1: ideapad_bluetooth: Bluetooth
	Soft blocked: no
	Hard blocked: no
2: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
[root@localhost rtl8723au_bt-kernel]# 

And on the user interface on gnome I see no bluetooth functioning. 
I saw issue 'bluetooth not working while wifi is on' so I stopped the wifi but got same result.

Thanks

lwfinger, 2015-06-15:
Reboot and run two commands:

lsmod | grep btusb
lsusb

Post the output of the two commands.

mehanzhi, 2015-06-16:
Hello,

[root@admin-riker-deltanews-bg ~]# uptime
 09:28:46 up 2 min,  2 users,  load average: 1.34, 0.44, 0.15
[root@admin-riker-deltanews-bg ~]# lsmod | grep btusb
[root@admin-riker-deltanews-bg ~]# lsusb
Bus 004 Device 004: ID 04f2:b322 Chicony Electronics Co., Ltd 
Bus 004 Device 003: ID 2047:0855 Texas Instruments Invensense Embedded MotionApp HID Sensor
Bus 004 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 005: ID 04f3:000a Elan Microelectronics Corp. Touchscreen
Bus 003 Device 004: ID 0bda:1724 Realtek Semiconductor Corp. RTL8723AU 802.11n WLAN Adapter
Bus 003 Device 003: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
Bus 003 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
[root@admin-riker-deltanews-bg ~]# dmesg |grep -i btusb
[root@admin-riker-deltanews-bg ~]# 

The module doesn't seem to be loaded on boot. But even if I load it manually it would still not work.

Thanks.

lwfinger, 2015-06-16:
When a module is not loaded on boot, that means the device is not in its tables. Forcing a load won't help either. That said, I do not understand your problem. The USB ID 0bda:1724 is in the version of the driver in the kernel branch. Are you sure you are using that branch? The command 'git branch' will tell you that.

mehanzhi, 2015-06-16:
What I did was 
go to https://github.com/lwfinger/rtl8723au_bt/, select 'Branches' -> Kernel branch.

Then download zip file. 
Decompress it and inside there was directory rtl8723au_bt-kernel

I suppose this means it's a kernel branch? 

Thanks

zhukovra, 2015-06-16:
@mehanzhi yes, it is kernel branch archive.

mehanzhi, 2015-06-16:
Thanks zhukovra,

lwfinger, would you like ssh access, logs, or something else in order to debug?

Thanks.

lwfinger, 2015-06-16:
As said earlier, you have the kernel branch, and it should work.

Send ssh access info to Larry.Finger@lwfinger.net.

noobermin, 2015-06-28:
I have a similar issue. Is there is progress on this, I'd appreciate an update on this thread.

mehanzhi, 2015-06-28:
Hello,
lwfinger is actively working on it and testing on my box. He can give you technical details.

lwfinger, 2015-06-28:
I finally got a chance to work on this, and pushed a trial fix onto the GitHub site. Unfortunately, I was unable to log into the test box. Until we fix my connection problem, please try this code.

noobermin, 2015-06-28:
So, I see progress. Now, it seems that the bluetooth is being detected, and hci0 shows up in hciconfig. However, `hciconfig hci0 up' outputs "Can't init device hci0: Invalid argument (22)". Also, dmesg shows an error: "Bluetooth: RTL8723A: unexpected EPATCH signature!"

See
http://pastebin.com/J8Widnv7

lwfinger, 2015-06-28:
That is progress. I just pushed new BT firmware for the rtl8723au. Please pull, do a 'sudo make install' and try again.

noobermin, 2015-06-28:
Cheers  @lwfinger, it works again. Thanks for everything!

mehanzhi, 2015-06-28:
Hello,

I was at work and I had to take my laptop with me, however it's now available for testing if you like. 

I'll try the code too and see if it will work for me.

mehanzhi, 2015-06-28:
Hmm, the module is loaded, but still can't find my phone when searching for devices I can't see any available devices :(

lwfinger, 2015-06-28:
I logged in, and updated the driver and firmware to the latest versions. I do not know if the correct firmware is loaded or not. The system will need to be rebooted. Unfortunately, the following messages are back:

atkbd serio0: Unknown key released (translated set 2, code 0xbe on isa0060/serio0).
atkbd serio0: Use 'setkeycodes e03e <keycode>' to make it known.

Those messages overrun all the useful stuff.

noobermin, 2015-06-28:
@mehanzhi that issue is an annoyance with the yoga 13. Basically, it's due to the one of the screen modes notifications not being understood by the kernel. The way I "fix" it is to go into windows and click the "lock mode" button (the small one on the front right side) and log back into linux and see if it leaves.

lwfinger, 2015-06-28:
If you run 'sudo crontab -e' and add a line with "@onboot /usr/bin/setkeycodes e03e", that should fix it.

lwfinger, 2015-06-29:
Sorry, that crontab change will not help.

mehanzhi, 2015-06-29:
Hello,

I fix it with the script 

#!/bin/sh

sudo setkeycodes e00d 255
sudo setkeycodes e03e 255

But I have throubles make the /etc/rc.local run so the script can be started on boot and I have to run it manually every time. This time I forgot to run it, sorry about that :(

And about the solution with windows, I don't have windows on my box. And the little button for screen rotation is nothing else than keys Supr+O and it sets a setting on/off in the gconf so you can use it to enable/disable screen rotation, I have a custom script for this too :)

mehanzhi, 2015-06-29:
Rebooted, driver is loaded and bluetooth device is available but still I can't find any bluetooth devices.

lwfinger, 2015-06-29:
Using "sudo crontab -e", add a line that says "@onboot <path to your script>" You should remove the "sudo" from the commands, and use the full path of /usr/bin/setketcodes. Also make sure that the script is executable.

I am unable to connect to the box.

mehanzhi, 2015-07-13:
Hello,

Issue with the bluetooth seems to be solved for me too.

Thanks for the help.

jordirelats, 2015-07-21:
Hello,

I have updated on opensuse tumbleweed to Kernel 4.1.1, I understood that should have the bluetooth driver on it, but it seems that is not there. I have installed rtl8723au_bt-kernel. Now, it seems that the bluetooth is being detected, and hci0 shows up in hciconfig. However, `hciconfig hci0 up' outputs "Can't init device hci0: Invalid argument (22)". Also, dmesg shows an error: "Bluetooth: RTL8723A: unexpected EPATCH signature!"

Any sugestion?

lwfinger, 2015-07-21:
The new driver will be in kernel 4.2. It is not in 4.1.

There is new firmware that was pushed yesterday. Did you pull again?

jordirelats, 2015-07-22:
I have download and install again just in case, still not working here some information:

linux-g7rn:/home/jordi # rfkill list
0: ideapad_wlan: Wireless LAN
        Soft blocked: no
        Hard blocked: no
1: ideapad_bluetooth: Bluetooth
        Soft blocked: no
        Hard blocked: no

linux-g7rn:/home/jordi # lsusb
Bus 002 Device 004: ID 04f2:b322 Chicony Electronics Co., Ltd 
Bus 002 Device 003: ID 2047:0855 Texas Instruments Invensense Embedded MotionApp HID Sensor
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 005: ID 04f3:000a Elan Microelectronics Corp. Touchscreen
Bus 001 Device 004: ID 0bda:1724 Realtek Semiconductor Corp. RTL8723AU 802.11n WLAN Adapter
Bus 001 Device 003: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
2: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no
3: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no

linux-g7rn:/home/jordi # lsmod | grep btusb
btusb                  53248  0 
bluetooth             528384  7 bnep,btusb

linux-g7rn:/home/jordi # hciconfig hci0 up
Can't init device hci0: Invalid argument (22)

linux-g7rn:/home/jordi # hciconfig list
hci0:   Type: BR/EDR  Bus: USB
        BD Address: 00:00:00:00:00:00  ACL MTU: 0:0  SCO MTU: 0:0
        DOWN 
        RX bytes:42 acl:0 sco:0 events:3 errors:0
        TX bytes:9 acl:0 sco:0 commands:3 errors:0

Thankyou for your help

jordirelats, 2015-07-22:
Just I mix up 
rfkill list
0: ideapad_wlan: Wireless LAN
        Soft blocked: no
        Hard blocked: no
1: ideapad_bluetooth: Bluetooth
        Soft blocked: no
        Hard blocked: no
2: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no
3: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no

jordirelats, 2015-07-24:
I just update to Kernel 4.1.2 and still the same problems

ivaniclixx, 2016-02-24:
root@tobor:~/rtl8723au_bt-kernel# hciconfig 
hci0:	Type: BR/EDR  Bus: USB
	BD Address: A0:F9:59:7F:F0:66  ACL MTU: 820:8  SCO MTU: 255:16
	UP RUNNING PSCAN ISCAN 
	RX bytes:643 acl:0 sco:0 events:42 errors:0
	TX bytes:1000 acl:0 sco:0 commands:41 errors:0

root@tobor:~/rtl8723au_bt-kernel# hcitool scan
Scanning ...

[ Never finds a device !! ] 

root@tobor:~/rtl8723au_bt-kernel# bluetoothctl 
[NEW] Controller A0:F9:59:7F:F0:66 tobor [default]
[bluetooth]# devices 
[bluetooth]#

root@tobor:~/rtl8723au_bt-kernel# uname -a
Linux tobor 4.1.13-v7+ #826 SMP PREEMPT Fri Nov 13 20:19:03 GMT 2015 armv7l GNU/Linux

With a Debian Jessie 3.16.0-4-amd64 it works perfectly.