lwfinger/rtl8723au_bt #34: Kernel panic with kernel 4.1+ when BT device enabled

ohgodwhyyy, 2015-07-27:
I am using Ubuntu 15.04. I have tried Mainline kernels 4.0 through 4.2 RC4 and with 4.1 and 4.2 BT is working out-of-the-box but I am getting kernel panics at seemingly random intervals when my BT mouse is turned on.

With kernel 4.0 I compiled wifi (rtlwifi_new-master) and BT (rtl8723au_bt-kernel) drivers and everything worked fine.

Compiling BT drivers on kernel 4.2 did not help, but as I understand, they are already included since this version. Compiling the wifi drivers failed on 4.2.

Any tips on how to debug the problem?

lwfinger, 2015-07-28:
The master branch of the wifi driver builds fine on my kernel 4.2-rc3; however, you should not need to use it with a kernel that new. The built-in driver should be fine.

I have no way to test the in-kernel BT driver as I do not have the hardware. If it does not work, or if it crashes, then you should report that to the mailing list at linux-bluetooth@vger.kernel.org.

krlk89, 2015-08-19:
Please close, this was a bug in some kernel releases. Seems to be fixed in 4.2 RC7.

closed 2015-08-19 by krlk89