lwfinger/rtl8723au_bt #36: rssi is always -90 dB for BLE devices

dlenski, 2015-08-15:
I've been using the `rtl8723au_bt/kernel` driver while hacking on a Bluetooth LE application.

I've noticed that `hci_read_rssi` *always* gives "-90 dB" as the [RSSI](https://en.wikipedia.org/wiki/Received_signal_strength_indication) using the `rtl8723au` adapter. Wireshark's packet dissection shows something like this for every response:

```
No.     Time               Source                Destination           Protocol Length Info
     29 20:28:31.416504000 controller            host                  HCI_EVT  10     Rcvd Command Complete (Read RSSI)

Frame 29: 10 bytes on wire (80 bits), 10 bytes captured (80 bits) on interface 0
Bluetooth HCI H4
    [Direction: Rcvd (0x01)]
    HCI Packet Type: HCI Event (0x04)
Bluetooth HCI Event - Command Complete
    Event Code: Command Complete (0x0e)
    Parameter Total Length: 7
    Number of Allowed Command Packets: 2
    Command Opcode: Read RSSI (0x1405)
        0001 01.. .... .... = Opcode Group Field: Status Parameters (0x0005)
        .... ..00 0000 0101 = Opcode Command Field: Read RSSI (0x0005)
    Status: Success (0x00)
    Connection Handle: 0x0010
    RSSI (dB): -90
```

At first I thought the `rtl8723au` firmware must just not be able to measure RSSI, but then I noticed that the LE advertising packets show a fluctuating range of RSSI values:

```
No.     Time               Source                Destination           Protocol Length Info
   1284 20:54:04.689747000 controller            host                  HCI_EVT  69     Rcvd LE Meta (LE Advertising Report)

Frame 1284: 69 bytes on wire (552 bits), 69 bytes captured (552 bits) on interface 0
Bluetooth Linux Monitor Transport
    [Adapter ID: 0]
    [Opcode: HCI Event Packet (0x0003)]
Bluetooth HCI Event - LE Meta
    Event Code: LE Meta (0x3e)
    Parameter Total Length: 67
    Sub Event: LE Advertising Report (0x02)
    Num Reports: 2
    Event Type: Connectable Unidirected Advertising (0x00)
    Peer Address Type: Random Device Address (0x01)
    BD_ADDR: TomtomSo_17:62:b1 (e4:04:39:17:62:b1)
    Data Length: 29
    Advertising Data
    RSSI (dB): -74
    Event Type: Scan Response (0x04)
    Peer Address Type: Random Device Address (0x01)
    BD_ADDR: TomtomSo_17:62:b1 (e4:04:39:17:62:b1)
    Data Length: 16
    Advertising Data
    RSSI (dB): -60
```

For comparison, I also tried a CSR8510 Bluetooth 4.0 dongle, and it does **not** have this deficiency (`hci_read_rssi` gives a range of values).

lwfinger, 2015-08-15:
I'm sorry, but I have too many real problems to work on trivial problems like this. As it seems to bother you, I suggest that you figure out what changes need to be made. That is the real value of open source.

closed 2015-08-15 by dlenski

dlenski, 2015-08-15:
Understood, sorry to take up your time. I noticed it because the peripheral I'm working with appears to have a very limited range. Seems likely to be a firmware issue so I'll play around with other firmware files in the repo.