lwfinger/rtl8723au_bt #37: Errors when make troy branch on Linux 4.2.0-1-MANJARO x86_64

rigzbe, 2015-09-04:
Hi (;
I have rtl8723be wifi and bt card. Wifi works ootb on 4.2 kernel, so I tried to install bt driver, but terminal gave me error messages. Output here → http://pastebin.com/NEvmhN9s

On kernel 316 wifi and bt didn`t work, I installed master branch from https://github.com/lwfinger/rtlwifi_new. Only wifi worked.
Then after reboot i installed kernel branch from https://github.com/lwfinger/rtl8723au_bt. Then i install troy branch of it and wifi and bt works good.

What I need to do to have bluetooth and wifi working on 4.20??

I can give all config files, outputs etc if need (;

lwfinger, 2015-09-04:
With kernel 4.2, you should be able to use the driver that comes in the kernel source. This repo is for older kernels where the BT driver does not support the Realtek devices. I will not be fixing the build failures you report.

Wifi should work in 4.2 with no problems. For BT, the only thing you need to do is make certain that file /lib/firmware/rtl_bt/rtl8723b_fw.bin exists. Your disto probably does not supply that yet. If the firmware is available, BT should work. If it does not, then you need to supply the appropriate part of the dmesg output that will describe the problem.

rigzbe, 2015-09-04:
Ok, I understand. Thanks for advice (-:
I checked, this file exist, but i got some unexplained errors in dmesg:
http://pastebin.com/JWa5eJT2

here is full dmesg:
http://pastebin.com/B7hAnwHr

rigzbe, 2015-09-05:
I think I figured it out, i just made changes in all of these files and wifi & bt works great on linux 4.2 :smile: 
I also replaced systemd with openrc :wink: 

/etc/rc.local:
hidd --connect <B0:10:41:A3:5D:xx>

/etc/conf.d/bluetooth:
# Arguments to hidd
HIDD_OPTIONS="--connect <B0:10:41:A3:5D:xx>"

/etc/udev/rules.d/10-local.rules
# Set bluetooth power up
ACTION=="add", KERNEL=="hci0", RUN+="/usr/bin/hciconfig hci0 up"

closed 2015-09-05 by rigzbe