lwfinger/rtl8723au_bt #39: Driver it seams is not in Kernel 4.2.1 opensuse

jordirelats, 2015-10-05:
I have installed the kernel 4.2.1 on opensuse tumbleweed, and the bluetooth adapter is not found.

linux-g7rn:/home/jordi # rfkill list
0: ideapad_wlan: Wireless LAN
        Soft blocked: no
        Hard blocked: no
1: ideapad_bluetooth: Bluetooth
        Soft blocked: no
        Hard blocked: no
2: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no

linux-g7rn:/home/jordi # lsusb
Bus 002 Device 004: ID 04f2:b322 Chicony Electronics Co., Ltd 
Bus 002 Device 003: ID 2047:0855 Texas Instruments Invensense Embedded MotionApp HID Sensor
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 005: ID 04f3:000a Elan Microelectronics Corp. Touchscreen
Bus 001 Device 004: ID 0bda:1724 Realtek Semiconductor Corp. RTL8723AU 802.11n WLAN Adapter
Bus 001 Device 003: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

I have try to install rtl8723au_bt apparently the installation is correct but the adapter is not found.

Jordi

stekman, 2015-10-09:
I have the same issue on Ubuntu 15.10 with kernel 4.2.0-15-generic #18-Ubuntu SMP  x86_64

lwfinger, 2015-10-09:
Please post the configuration of the kernel in some pastebin location. Your distro may have the configuration wrong.

jordirelats, 2015-10-10:
Sorry but I don't know how to do that, I'm only a linux user.

lwfinger, 2015-10-10:
What does being a Linux user have to do with that?

Your distro may put the configuration file in /boot/config-xxxx or it may be found in /proc/config.gz. Post either of those at http://pastebin.com and post the link here.

Without that info, I can do nothing for you.

jordirelats, 2015-10-10:
Thank you for your help.

This is the link of the kernel configuration http://pastebin.com/WK9uHYf1

Jordi

lwfinger, 2015-10-10:
The configuration looks correct. Using the standard driver in the kernel, please reboot and post the output of the dmesg command in a pastebin document. I need to see the LMP value for your device.

jordirelats, 2015-10-10:
here you have the dmesg

http://pastebin.com/cYdCeeYz

Jordi

lwfinger, 2015-10-11:
I see no attempt to load the BT. For example, I see the following in my dmesg

[114479.494040] Bluetooth: hci0: rtl: examining hci_ver=06 hci_rev=000a lmp_ver=06 lmp_subver=8821
[114479.494046] Bluetooth: hci0: rtl: loading rtl_bt/rtl8821a_fw.bin
[114479.494535] usbcore: registered new interface driver btusb
[114479.501081] Bluetooth: hci0: rom_version status=0 version=1
[114479.513040] Bluetooth: hci1: read Intel version: 370710018002030d00
[114479.513555] bluetooth hci1: Direct firmware load for intel/ibt-hw-37.7.10-fw-1.80.2.3.d.bseq failed with error -2
[114479.513562] Bluetooth: hci1 failed to open Intel firmware file: intel/ibt-hw-37.7.10-fw-1.80.2.3.d.bseq(-2)
[114479.513695] bluetooth hci1: Direct firmware load for intel/ibt-hw-37.7.bseq failed with error -2
[114479.513699] Bluetooth: hci1 failed to open default Intel fw file: intel/ibt-hw-37.7.bseq
[114479.640085] Bluetooth: RFCOMM TTY layer initialized
[114479.640106] Bluetooth: RFCOMM socket layer initialized
[114479.640127] Bluetooth: RFCOMM ver 1.11

Some of that is from the Intel device, but other is from the 8821ae device that I have.

What does 'lsmod | grep btusb' show any output?

Does 'sudo modprobe -v btusb' produce ant output?

stekman, 2015-10-11:
Same things on Ubuntu 15.10:

Ubuntu 15.10 latest kernel config: http://pastebin.com/t4A3aSk4
dmesg output (including modprobe at from terminal after login): http://pastebin.com/1Ck8UGvM
Output from some commands: http://pastebin.com/SZ1i7x45

jordirelats, 2015-10-12:
I have donne  'lsmod | grep btusb' and I get no output

Then I do 'sudo modprobe -v btusb' and I get:

jordi@linux-g7rn:~> sudo modprobe -v btusb
root's password:
insmod /lib/modules/4.2.1-1-desktop/kernel/drivers/bluetooth/btintel.ko 
insmod /lib/modules/4.2.1-1-desktop/kernel/drivers/bluetooth/btbcm.ko 
insmod /lib/modules/4.2.1-1-desktop/kernel/drivers/bluetooth/btrtl.ko 
insmod /lib/modules/4.2.1-1-desktop/kernel/drivers/bluetooth/btusb.ko reset=1

then I do again a 'lsmod | grep btusb' and I get:

jordi@linux-g7rn:~> lsmod | grep btusb
btusb                  45056  0 
btrtl                  16384  1 btusb
btbcm                  16384  1 btusb
btintel                16384  1 btusb
bluetooth             520192  9 bnep,btbcm,btrtl,btusb,btintel

lwfinger, 2015-10-12:
It is clear that btusb and friends are not being loaded automatically, but I do not know why. The device tables in the 4.2 kernel definitely say to probe all devices with vendor 0x0bda and class 0xe0, i.e. Realtek BT devices.

Please post the output of 'lsusb -v -d 0bda:'.

jordirelats, 2015-10-12:
Here you have, thank you for your help

jordi@linux-g7rn:~> lsusb -v -d 0bda:

Bus 001 Device 004: ID 0bda:1724 Realtek Semiconductor Corp. RTL8723AU 802.11n WLAN Adapter
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0x1724 RTL8723AU 802.11n WLAN Adapter
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          222
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              500mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass        224 Wireless
      bFunctionSubClass       1 Radio Frequency
      bFunctionProtocol       1 Bluetooth
      iFunction               4 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       224 Wireless
      bInterfaceSubClass      1 Radio Frequency
      bInterfaceProtocol      1 Bluetooth
      iInterface              4 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
[... 285 lines cut ...]
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0003  1x 3 bytes
        bInterval              10

stekman, 2015-10-12:
lsusb -v -d 0bda: 

gives almost identical output on my Ubuntu kernel. With one tiny difference. A question mark on the first:

bDeviceSubClass         2 ?

lwfinger, 2015-10-13:
I do not know what is wrong. The device table command says to accept any device with class 224, subclass 1, and protocol 1. Your devices match that requirement.

I just wrote a RedHat developer that has this device. Perhaps he has some special insight.

MilkEnd, 2015-10-16:
In kernel 4.2.3 source tree, devices with 0bda vendor id are not listed in btusb_table[]  but in blacklist_table[] in file btusb.c

tapiama, 2015-10-30:
thanks.  I moved 0bda line over to the btusb_table[] and it's all good now!

jordirelats, 2015-10-30:
You can explain how to do that? I'm not linux expert and I don't kow how to do it

flortsch, 2015-10-30:
Hello guys,

I also got the problem, that my device couldn't be detected. I edited the btusb.c file and moved the realtek entries from the blacklist table into the btusb_table, compiled and installed. After a reboot bluetooth works fine.

@jordirelats 
You simply have to clone the repo and edit the btusb.c file within the repo.
Search for the following line:
static const struct usb_device_id blacklist_table[] = {

Below that, you have look for the realtek entries. Delete the realtek entries there and insert them after the following line:
static const struct usb_device_id btusb_table[] = {

Save the file, run "make" and "sudo make install" in the repo directory and reboot.

stekman, 2015-11-01:
Thanks a lot! It worked!

jordirelats, 2015-11-06:
Hello

Sorry, but what you mention to me is not so easy, I'm just a linux user.

Please you can explain how to do this?

Thank you very much

jordirelats, 2015-11-07:
Hello,

This issue will be fixed on kernel 4.3?

lwfinger, 2015-11-07:
No, it will not be fixed in 4.3 as the fix was found too late. I just pushed a fix to the kernel branch of this repo. The driver now works for RTL8723BU and should work for RTL8723AU, although I have no hardware for testing.

lwfinger, 2015-11-07:
The kernel branch of the repo now has this fix. It is needed for both RTL8723AU and RTL8723BU. The "unique" feature for these two devices is that they use the same USB ID for both wifi and BlueTooth.

stekman, 2015-11-08:
I have tested the kernel branch with rtl8723au and can confirm that it works. This is on a Lenovo Yoga with Ubuntu 15.10 with latest kernel updated from ubuntu repos (4.2.0-17).

jordirelats, 2015-11-08:
I have tested the kernel branch with rtl8723au and can confirm that it works. This is on a Lenovo Yoga with OpenSuse Tumbleweed with Kernel 4.3.0

Do you know in witch Kernel will be solved the issue?

dmnk, 2015-11-20:
hi,
i'm using suse leap with the 4.1.12-1-default kernel and a rtl8723be card.

```
lsmod | grep btusb
btusb                  45056  0 
btbcm                  16384  1 btusb
btintel                16384  1 btusb
bluetooth             532480  38 bnep,btbcm,btusb,rfcomm,btintel
usbcore               262144  7 btusb,uvcvideo,ehci_hcd,ehci_pci,usbhid,xhci_hcd,xhci_pci
```

```
 lsusb -v -d 0bda: 

Bus 001 Device 003: ID 0bda:b001 Realtek Semiconductor Corp. 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass          224 Wireless
  bDeviceSubClass         1 Radio Frequency
  bDeviceProtocol         1 Bluetooth
  bMaxPacketSize0        64
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0xb001 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          177
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       224 Wireless
      bInterfaceSubClass      1 Radio Frequency
      bInterfaceProtocol      1 Bluetooth
      iInterface              4 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
[... 232 lines cut ...]
        Name: linux-XX
        Alias: linux-XX
        Class: 0x1c010c
        Powered: yes
        Discoverable: yes
        Pairable: yes
        UUID: Headset AG                (00001112-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: OBEX File Transfer        (00001106-0000-1000-8000-00805f9b34fb)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: OBEX Object Push          (00001105-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: IrMC Sync                 (00001104-0000-1000-8000-00805f9b34fb)
        UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: Message Notification Se.. (00001133-0000-1000-8000-00805f9b34fb)
        UUID: Phonebook Access Server   (0000112f-0000-1000-8000-00805f9b34fb)
        UUID: Message Access Server     (00001132-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v1D6Bp0246d0523
        Discovering: yes
```

but the scan doesn't discover any devices.

leon9923, 2016-09-07:
@dmnk did you find a fix for it after that I'm also stuck here

dmnk, 2016-09-08:
@leon9923 kind of - the hard way: i've bought a intel wifi card.
as direct replacement the 3160 or if you take this as a chance to upgrade (2x2 instead 1x1 as the 3160 and rtl8723, and AC capable): 7260

sorry, for no better solution, but i should have changed immediately to an intel card, instead of wasting time with realtek. right now they are between 16 und 22 € on amazon