lwfinger/rtl8723au_bt #40: Tweaks for Toshiba Satellite L15W-B (Radius)

xterminus, 2015-10-17:
Been fooling with these drivers trying to get them working on my (newish) laptop.  bluetooth device always showed up as:
```
0: Toshiba Bluetooth: Bluetooth
	Soft blocked: no
	Hard blocked: no
1: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
3: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
```
and poking it with lsusb showed the device as

    Bus 001 Device 010: ID 0930:0222 Toshiba Corp.

prodding it for more info returned the following.  Nowhere did it mention realtek chipset so I figured maybe it wasn't being recognized properly and maybe Toshiba had a custom version made for this line of laptops:
```
Bus 001 Device 010: ID 0930:0222 Toshiba Corp. 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass          224 Wireless
  bDeviceSubClass         1 Radio Frequency
  bDeviceProtocol         1 Bluetooth
  bMaxPacketSize0        64
  idVendor           0x0930 Toshiba Corp.
  idProduct          0x0222 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          177
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       224 Wireless
      bInterfaceSubClass      1 Radio Frequency
      bInterfaceProtocol      1 Bluetooth
      iInterface              4 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
[... 202 lines cut ...]
    /* Additional Realtek 8723AE Bluetooth devices */
    { USB_DEVICE(0x0930, 0x021d), .driver_info = BTUSB_REALTEK },
    { USB_DEVICE(0x13d3, 0x3394), .driver_info = BTUSB_REALTEK },

New:

    /* Additional Realtek 8723AE Bluetooth devices */
    { USB_DEVICE(0x0930, 0x021d), .driver_info = BTUSB_REALTEK },
    { USB_DEVICE(0x0930, 0x0222), .driver_info = BTUSB_REALTEK },
    { USB_DEVICE(0x13d3, 0x3394), .driver_info = BTUSB_REALTEK },

After recompiling and re-inserting the module, dmesg showed:
```
[    6.000307] Bluetooth: hci0: rtl: examining hci_ver=06 hci_rev=000b lmp_ver=06 lmp_subver=8723
[    6.000314] Bluetooth: hci0: rtl: loading rtl_bt/rtl8723b_fw.bin
```
So I assume that means it's loading the firmware now.  And bluetooth works.  So yay!

Anyway, thought I'd share my success and hope someone else finds this useful.
```
System:    Kernel: 4.2.0-16-generic x86_64 (64 bit) Desktop: Gnome 3.18.0 Distro: Ubuntu 15.10 wily
Machine:   System: TOSHIBA product: Satellite L15W-B v: PSKVQU-002002
           Mobo: TOSHIBA model: MA20 Bios: TOSHIBA v: 1.30 date: 05/22/2015
Network:   Card: Realtek RTL8723BE PCIe Wireless Network Adapter driver: rtl8723be
```

tapiama, 2015-10-30:
Thanks for sharing your experience.  I have a different laptop (Yoga 13) with different config:

Bus 003 Device 004: ID 0bda:1724 Realtek Semiconductor Corp. RTL8723AU 802.11n WLAN Adapter

I read your post and thought I'd try that - instead of adding new entry 0bda:1724 to the blacklist, I moved the generic realtek line from blacklist to the btusb_table[] array

{ USB_VENDOR_AND_INTERFACE_INFO(0x0bda, 0xe0, 0x01, 0x01), .driver_info = BTUSB_REALTEK },

I've already used lwfinger's modules, but after upgraded to 15.10 I had to use the "kernel" branch and above modification to make it work.  There are some differences between "kernel" branch's btusb.c and "new" branch's btusb.c.  If you device used to work by "new" branch, perhaps this is a place to check.  Good luck.

tapiama, 2015-10-30:
BTW, once bluetooth is up, under 15.10 I was able to pair and use my Bluetooth 4.0 (low-energy profile)  micosoft designer mouse right from standard UI.  Cool..  And both wifi/bluetooth works fine simultaneously with lwfinger's drivers (there was a question about that in another thread)

N4RPS, 2015-12-15:
Hello!

I'm having the same problem you did. Is there any way this can be passed upstream, to be included in future updates? Having to patch this file every time it's updated would be a real bummer...

lwfinger, 2015-12-15:
The patch has been sent upstream, but I have heard nothing since.

N4RPS, 2015-12-16:
Hello!

I finally found what I needed on the Ubuntu forums. Bless the soul who created a PPA to handle this issue as new kernels arrive!

73 DE N4RPS
Rob

mochnaufals, 2021-05-15:
Sorry for commenting on an old issue.

But I wonder if the line `{ USB_DEVICE(0x0930, 0x0222), .driver_info = BTUSB_REALTEK },` has been added to the newer kernel, since my Bluetooth Realtek ID is 0930:0222 and I still have Bluetooth problem with kernel 5.10.34-1-MANJARO. I checked this link https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/bluetooth/btusb.c?h=v5.12.4 and I did not found that line.

Any suggestions of what should I do?
Many thanks.

lwfinger, 2021-05-15:
The USB vendor ID of 0x0930 belongs to Atheros, not Realtek. Are you sure that your BT device is made by Realtek?

mochnaufals, 2021-05-15:
@lwfinger 
Thank you for replying. Yes, I'm pretty sure that my BT device is made by realtek. This is a picture of the back of my laptop,
[image]

I did not do any replacement to the bluetooth/WiFi hardware since I bought it, and one of my `lsusb` output returns `Bus 002 Device 004: ID 0930:0222 Toshiba Corp. Bluetooth Radio`

blisshome, 2022-09-14:
I'm so glad your post from 2015 was still up. Your solution worked perfectly! On the other hand, doing my first kernel recompile was, let's just say "a learning experience". I hope you see this post and feel my wireless gratitude.

lwfinger, 2022-09-14:
I remember my first kernel build, but the kernel was a lot smaller. You are welcome.

SonoDavid, 2023-10-15:
Compiling now. Is there any possibility for this change to be included in future linux kernels?