lwfinger/rtl8723au_bt #41: Not 'make' on 3.18 kernel branch

rigzbe, 2015-11-11:
Hai :hand:  , today I wanted to update drivers. I downloaded kernel branch, unpacked, cd and typed make.
Make made errors and output:

`https://pad.riseup.net/p/Oai25KpPZuaa'

How to fix it?

lwfinger, 2015-11-11:
The kernel branch is for current kernels, that is 4.2 or later. If you want to build on 3.18, you may be able to use one of the other branches. For 3.18, you would need to copy a lot of code from these later kernels.

rigzbe, 2015-11-11:
I built for 4.2.5-1 with no errors. Before and after install my device is not discoverable and unable to find other devices.

lwfinger, 2015-11-11:
It works for me with BT part of RTL8723BU and kernel 4.3.

rigzbe, 2015-11-11:
Ok, I'm installing 43 kernel now and later I'll post about it.
I have device  RTL8723BE 0bda:b728

PS [localhost kuba]# rfkill list
0: ideapad_wlan: Wireless LAN
	Soft blocked: no
	Hard blocked: no
1: ideapad_bluetooth: Bluetooth
	Soft blocked: no
	Hard blocked: no
2: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
3: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no

If it is important :smile:

lwfinger, 2015-11-11:
The BE part will work with the BT driver in kernel 4.3. You do not need to use this repo. You do have to be careful with the btusb configuration.

closed 2015-11-11 by lwfinger