lwfinger/rtl8723au_bt #43: rtl8723au bluetooth device found, lescan finds one result and stops?

qmnonic, 2015-11-16:
**Summary**
I'm able to build the kernel branch on a raspberry pi b+ (jessie), install the drivers, view the devices and start an LE scan.  However, the LE scan only returns 1 result for the rtl8723bu device, whereas it returns dozens of results for a CSR bluetooth device.

**Hardware summary**
```
pi@raspberrypi ~/rtl8723au_bt $ uname -a
Linux raspberrypi 4.1.7+ #817 PREEMPT Sat Sep 19 15:25:36 BST 2015 armv6l GNU/Linux
pi@raspberrypi ~/rtl8723au_bt $ lsusb
Bus 001 Device 008: ID 0bda:b720 Realtek Semiconductor Corp.
Bus 001 Device 005: ID 0bda:8176 Realtek Semiconductor Corp. RTL8188CUS 802.11n WLAN Adapter
Bus 001 Device 004: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

**Building the drivers**
```
pi@raspberrypi ~/rtl8723au_bt $ date
Mon Nov 16 14:26:09 UTC 2015
pi@raspberrypi ~/rtl8723au_bt $ git pull
Already up-to-date.
pi@raspberrypi ~/rtl8723au_bt $ git status
On branch kernel
Your branch is up-to-date with 'origin/kernel'.
nothing to commit, working directory clean
pi@raspberrypi ~/rtl8723au_bt $
pi@raspberrypi ~/rtl8723au_bt $ make clean
rm -rf *.o *.mod.c *.mod.o *.ko *.symvers *.order *.a
pi@raspberrypi ~/rtl8723au_bt $
pi@raspberrypi ~/rtl8723au_bt $ make
make -C /lib/modules/4.1.7+/build M=/home/pi/rtl8723au_bt modules
make[1]: Entering directory '/home/pi/linux-59e76bb7e2936acd74938bb385f0884e34b91d72'
  CC [M]  /home/pi/rtl8723au_bt/btusb.o
  CC [M]  /home/pi/rtl8723au_bt/btrtl.o
  CC [M]  /home/pi/rtl8723au_bt/btintel.o
  CC [M]  /home/pi/rtl8723au_bt/btbcm.o
  Building modules, stage 2.
  MODPOST 4 modules
  CC      /home/pi/rtl8723au_bt/btbcm.mod.o
  LD [M]  /home/pi/rtl8723au_bt/btbcm.ko
  CC      /home/pi/rtl8723au_bt/btintel.mod.o
  LD [M]  /home/pi/rtl8723au_bt/btintel.ko
  CC      /home/pi/rtl8723au_bt/btrtl.mod.o
  LD [M]  /home/pi/rtl8723au_bt/btrtl.ko
  CC      /home/pi/rtl8723au_bt/btusb.mod.o
  LD [M]  /home/pi/rtl8723au_bt/btusb.ko
make[1]: Leaving directory '/home/pi/linux-59e76bb7e2936acd74938bb385f0884e34b91d72'
pi@raspberrypi ~/rtl8723au_bt $
pi@raspberrypi ~/rtl8723au_bt $ sudo make install
depmod -a /lib/modules/4.1.7+
installed revised btusb
pi@raspberrypi ~/rtl8723au_bt $
```

**Installing drivers**
```
pi@raspberrypi ~/rtl8723au_bt $ sudo modprobe -rv btusb
rmmod btusb
rmmod btintel
rmmod btbcm
pi@raspberrypi ~/rtl8723au_bt $
pi@raspberrypi ~/rtl8723au_bt $ sudo modprobe -v btusb
insmod /lib/modules/4.1.7+/kernel/drivers/bluetooth/btintel.ko
insmod /lib/modules/4.1.7+/kernel/drivers/bluetooth/btbcm.ko
insmod /lib/modules/4.1.7+/kernel/drivers/bluetooth/btusb.ko
pi@raspberrypi ~/rtl8723au_bt $
pi@raspberrypi ~/rtl8723au_bt $ dmesg | tail
[37141.874215] Bluetooth: hci0 urb d6b08600 failed to resubmit (2)
[37145.824820] usbcore: registered new interface driver btusb
[37579.022835] usbcore: deregistering interface driver btusb
[37579.023067] Bluetooth: hci1 urb d6b08600 failed to resubmit (2)
[37579.023164] Bluetooth: hci1 urb d6b08180 failed to resubmit (2)
[37579.023211] Bluetooth: hci1 urb d6b08500 failed to resubmit (2)
[37579.040417] Bluetooth: hci0 urb d6b08980 failed to resubmit (2)
[37579.040488] Bluetooth: hci0 urb d6b08f00 failed to resubmit (2)
[37579.040531] Bluetooth: hci0 urb d6b08e00 failed to resubmit (2)
[37583.504172] usbcore: registered new interface driver btusb
pi@raspberrypi ~/rtl8723au_bt $
```

**Device details via bluez post driver install**
```
pi@raspberrypi ~/rtl8723au_bt $ hciconfig -a
hci1: Type: BR/EDR  Bus: USB
   BD Address: 5C:F3:70:24:3E:BC  ACL MTU: 820:8  SCO MTU: 255:16
   UP RUNNING
   RX bytes:610 acl:0 sco:0 events:36 errors:0
   TX bytes:942 acl:0 sco:0 commands:36 errors:0
   Features: 0xff 0xff 0xff 0xfe 0xdb 0xff 0x7b 0x87
   Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3
   Link policy: RSWITCH HOLD SNIFF PARK
   Link mode: SLAVE ACCEPT
   Name: 'raspberrypi'
   Class: 0x000000
   Service Classes: Unspecified
   Device Class: Miscellaneous,
   HCI Version: 4.0 (0x6)  Revision: 0xe2f
   LMP Version: 4.0 (0x6)  Subversion: 0x9f73
   Manufacturer: Realtek Semiconductor Corporation (93)

hci0: Type: BR/EDR  Bus: USB
   BD Address: 00:15:83:D1:06:D1  ACL MTU: 310:10  SCO MTU: 64:8
   UP RUNNING
   RX bytes:610 acl:0 sco:0 events:36 errors:0
   TX bytes:942 acl:0 sco:0 commands:36 errors:0
   Features: 0xff 0xff 0x8f 0xfe 0xdb 0xff 0x5b 0x87
   Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3
   Link policy: RSWITCH HOLD SNIFF PARK
   Link mode: SLAVE ACCEPT
   Name: 'raspberrypi #1'
   Class: 0x000000
   Service Classes: Unspecified
   Device Class: Miscellaneous,
   HCI Version: 4.0 (0x6)  Revision: 0x22bb
   LMP Version: 4.0 (0x6)  Subversion: 0x22bb
   Manufacturer: Cambridge Silicon Radio (10)
```

**LEScan attempt for rtl8723bu device returns one result and stops**
```
pi@raspberrypi ~/rtl8723au_bt $ cat /proc/uptime
37662.55 35221.92
pi@raspberrypi ~/rtl8723au_bt $ sudo hcitool -i hci1 lescan
LE Scan ...
E2:72:03:D3:3B:B7 (unknown)
^Cpi@raspberrypi ~/rtl8723au_bt $ cat /proc/uptime
37684.37 35243.46
```

**LEScan attempt for CSR bluetooth device returns results quickly and continues forever**
```
pi@raspberrypi ~/rtl8723au_bt $ cat /proc/uptime
37684.37 35243.46
pi@raspberrypi ~/rtl8723au_bt $
pi@raspberrypi ~/rtl8723au_bt $ sudo hcitool -i hci0 lescan
LE Scan ...
7C:D1:C3:3B:6A:17 (unknown)
E2:72:03:D3:3B:B7 (unknown)
E2:72:03:D3:3B:B7 estimote
7C:D1:C3:3B:6A:17 (unknown)
7C:D1:C3:3B:6A:17 (unknown)
7C:D1:C3:3B:6A:17 (unknown)
7C:D1:C3:3B:6A:17 (unknown)
7C:D1:C3:3B:6A:17 (unknown)
7C:D1:C3:3B:6A:17 (unknown)
7C:D1:C3:3B:6A:17 (unknown)
E9:14:2B:53:67:5B (unknown)
E9:14:2B:53:67:5B estimote
7C:D1:C3:3B:6A:17 (unknown)
7C:D1:C3:3B:6A:17 (unknown)
7C:D1:C3:3B:6A:17 (unknown)
7C:D1:C3:3B:6A:17 (unknown)
^Cpi@raspberrypi ~/rtl8723au_bt $ cat /proc/uptime
37693.06 35251.87
pi@raspberrypi ~/rtl8723au_bt $
```

I tried all the other branches, but the kernel branch had the best results.  I don't see any errors.  Not sure what's going on.

lwfinger, 2015-11-16:
I'm not sure either. I do not have an RTL8723AU device, but I have a number of other Realtek devices with associated BT hardware. All of them find all the BT devices that I have here as long as I use "scan" in the hcitool command. When I try the lescan, nothing responds. I'm not sure why, but LE connections appear not to be possible. For what it is worth, the same is true for an Intel BT device as well.

qmnonic, 2015-11-16:
hw correction: RTL8723BU device.

Strange that the le scan wouldn't work.
I'm going to try to update bluez.
Might also try poking around in the source (but won't hold out hope).

lwfinger, 2015-11-16:
I do have a BU device. Same situation - scan works, but LE scan does not.

qmnonic, 2015-11-16:
updated bluez doesn't seem to make a difference in this case.

qmnonic, 2015-11-18:
'new' branch on kernel 3.1.8.7+ seems to scan a few more devices.  Interestingly, they are unique.

```
pi@raspberrypi ~ $ uname -a
Linux raspberrypi 3.18.7+ #755 PREEMPT Thu Feb 12 17:14:31 GMT 2015 armv6l GNU/Linux
```

scan from rtl8723bu:

```
pi@raspberrypi ~/rtl8723au_bt $ sudo hcitool -i hci1 lescan
LE Scan ...
7C:D1:C3:3B:6A:17 (unknown)
E2:72:03:D3:3B:B7 (unknown)
E9:14:2B:53:67:5B (unknown)
```

qmnonic, 2015-11-18:
Also interesting - LE advertising appears to work (kernel 3.18.7+ so far).

qmnonic, 2015-11-22:
'test' branch works for LE advertising on kernel 4.1.7+