lwfinger/rtl8723au_bt #44: error compiling on linux-headers-4.2.0-19-generic

kubwit, 2015-11-21:
Makefile:15: set of instructions for "all" not possibel
make: *** [all] Errore 2

This is the error i get compiling rtl8723au_bt on 4.2.0-19 Kernel Ubuntu 15.10 Gnome

Maybe can be useful

lwfinger, 2015-11-21:
That is only the top-level error from make. It only says that something happened earlier. Without the previous messages, there is nothing I can fix.

Be aware that Ubuntu messes with driver APIs in a method that I refuse to support as that makes building on Ubuntu different than other distros. If that is the reason for your problem, you will be on your own. Sorry.

kubwit, 2015-11-21:
thanks for your reply. i only wanted to contribute. I have a general problem on my laptop with the original drivers. Bluetooth only connects like a headphone with my phone and wireless is only connecting for webservices but fileservices like nfs ssh ecc don't work. only for your knowledge
i'm on an lenovo g50-70 and before upgrading i used your drivers always and they worked perfect.

see this topic http://forums.linuxmint.com/viewtopic.php?f=49&t=193838
02:00.0 Network controller [0280]: Realtek Semiconductor Co., Ltd. RTL8723BE PCIe Wireless Network Adapter [10ec:b723]
   Subsystem: Lenovo Device [17aa:b736]
   Kernel driver in use: rtl8723be

Here are the other messages:
andreas@andreas-laptop:~/rtl8723au_bt$ make
make -C /lib/modules/4.2.0-19-generic/build M=/home/andreas/rtl8723au_bt modules
make[1]: ingresso nella directory "/usr/src/linux-headers-4.2.0-19-generic"
  CC [M]  /home/andreas/rtl8723au_bt/btusb.o
/home/andreas/rtl8723au_bt/btusb.c: In function ‘hci_reassembly’:
/home/andreas/rtl8723au_bt/btusb.c:111:14: error: ‘NUM_REASSEMBLY’ undeclared (first use in this function)
     index >= NUM_REASSEMBLY)
              ^
/home/andreas/rtl8723au_bt/btusb.c:111:14: note: each undeclared identifier is reported only once for each function it appears in
/home/andreas/rtl8723au_bt/btusb.c:114:12: error: ‘struct hci_dev’ has no member named ‘reassembly’
  skb = hdev->reassembly[index];
            ^
/home/andreas/rtl8723au_bt/btusb.c:141:7: error: ‘struct hci_dev’ has no member named ‘reassembly’
   hdev->reassembly[index] = skb;
       ^
/home/andreas/rtl8723au_bt/btusb.c:163:10: error: ‘struct hci_dev’ has no member named ‘reassembly’
      hdev->reassembly[index] = NULL;
          ^
/home/andreas/rtl8723au_bt/btusb.c:176:10: error: ‘struct hci_dev’ has no member named ‘reassembly’
      hdev->reassembly[index] = NULL;
          ^
/home/andreas/rtl8723au_bt/btusb.c:189:10: error: ‘struct hci_dev’ has no member named ‘reassembly’
      hdev->reassembly[index] = NULL;
          ^
/home/andreas/rtl8723au_bt/btusb.c:207:8: error: ‘struct hci_dev’ has no member named ‘reassembly’
    hdev->reassembly[index] = NULL;
        ^
/home/andreas/rtl8723au_bt/btusb.c: In function ‘btusb_close’:
/home/andreas/rtl8723au_bt/btusb.c:649:18: error: ‘NUM_REASSEMBLY’ undeclared (first use in this function)
  for (i = 0; i < NUM_REASSEMBLY; i++)
                  ^
/home/andreas/rtl8723au_bt/btusb.c:651:11: error: ‘struct hci_dev’ has no member named ‘reassembly’
   if (hdev->reassembly[i])
           ^
/home/andreas/rtl8723au_bt/btusb.c:653:18: error: ‘struct hci_dev’ has no member named ‘reassembly’
    kfree_skb(hdev->reassembly[i]);
                  ^
/home/andreas/rtl8723au_bt/btusb.c:654:8: error: ‘struct hci_dev’ has no member named ‘reassembly’
    hdev->reassembly[i] = NULL;
        ^
scripts/Makefile.build:264: set di istruzioni per l'obiettivo "/home/andreas/rtl8723au_bt/btusb.o" non riuscito
make[2]: *** [/home/andreas/rtl8723au_bt/btusb.o] Errore 1
Makefile:1398: set di istruzioni per l'obiettivo "_module_/home/andreas/rtl8723au_bt" non riuscito
make[1]: *** [_module_/home/andreas/rtl8723au_bt] Errore 2
make[1]: uscita dalla directory "/usr/src/linux-headers-4.2.0-19-generic"
Makefile:15: set di istruzioni per l'obiettivo "all" non riuscito
make: *** [all] Errore 2
andreas@andreas-laptop:~/rtl8723au_bt$ 

I hope this helps for your work. Thanks

lwfinger, 2015-11-21:
It tells me what is wrong. For kernel 4.2, you need to use the kernel branch. The code in the other branches is for older kernels. As the kernel driver now works, there is no need to port those older branches.

kubwit, 2015-11-21:
ok . thanks for your work.

dennohpeter, 2018-11-19:
> It tells me what is wrong. For kernel 4.2, you need to use the kernel branch. The code in the other branches is for older kernels. As the kernel driver now works, there is no need to port those older branches.

Yeah That worked fine thanks for your work.