lwfinger/rtl8723au_bt #46: Not able to share internet conection with android phone.

Lumumba, 2015-12-17:
no bnep0 device created no networking possible.
Works with some other bluetooth stick i have.

lwfinger, 2015-12-17:
How am I supposed to help when you give so little information?

Lumumba, 2015-12-17:
Sorry I am quite a kinux noob. I just used this driver like someone said it to me. And evreything works just fine. Exept i cannot share my Internet connection from my phone. I did some research but my linux knowledge is quite limited. 
And then I tried another bluetooth stick and a live image of my distribution and the everything worked fine.
So i guessed it is a driver issue but I know I could be wrong.
The driver I used is rtl8723au_bt-new.
I have a notebook with built in wifi and bluetooth. 
I asked in some Irc channels for the setup, I am happy it runs as much as it does now thanks to you.
I really do not have any Idea how this all works. 
But thanks for your help.

Lumumba, 2015-12-17:
And sorry for waisting your time.

lwfinger, 2015-12-17:
That is no problem, but you should be a little more complete is the initial description of the problem.

Sorry, but as ordinary wireless access points are so cheap, I concentrate on making the wireless drivers work as clients, the BT drivers to support my test mouse, and spend little time worrying about using the devices in the manner you suggest.

closed 2015-12-17 by lwfinger

Lumumba, 2015-12-17:
Ah my english is not the best i meant using the internet connection of the phone with the PC.