lwfinger/rtl8723au_bt #48: Support for kernel 4.4

gssci, 2016-04-05:
Greetings, I wonder - are you planning an update for kernel 4.4 for the release of the new version of Ubuntu? I'm running 16.04 beta 2 right now, and I'm debating if I should downgrade or just wait a little longer

Thank you very much for your work, you saved my new machine from an otherwise unavoidable windows configuration

nikolaandj, 2016-04-14:
I can confirm that this is a problem. `make` does work, while `sudo make install` throws the following error:

```
depmod -a /lib/modules/4.4.6-301.fc23.x86_64
depmod: ERROR: Bad version passed /lib/modules/4.4.6-301.fc23.x86_64
Makefile:32: recipe for target 'install' failed
make: *** [install] Error 1
```

To be clear, I am using _Fedora 23_ with _4.4.6-301_ kernel, and I am trying to compile the _kenel_ branch. The device is RTL8723BE on HP Probook 455 G2. Thank you in advance.

lwfinger, 2016-04-14:
The "kernel_version" branch is only intended to be built using the current stable kernel. At the moment, that is 4.5.0, not 4.4.X. You should be using the master branch.

nikolaandj, 2016-04-15:
Which branch or commit do you recommend for this kernel version?

nikolaandj, 2016-04-15:
I switched to the main branch and the same thing happends. `make` passes, while the `sudo make install` gives the following error:

```
depmod -a /lib/modules/4.4.6-301.fc23.x86_64
depmod: ERROR: Bad version passed /lib/modules/4.4.6-301.fc23.x86_64
Makefile:32: recipe for target 'install' failed
make: *** [install] Error 1
```

nikolaandj, 2016-04-15:
I switched to the master branch. This time the `make` fails with the following error:

```
make -C /lib/modules/4.4.6-301.fc23.x86_64/build M=/home/nikola/Git/rtl8723au_bt modules
make[1]: Entering directory '/usr/src/kernels/4.4.6-301.fc23.x86_64'
  CC [M]  /home/nikola/Git/rtl8723au_bt/rtk_btusb.o
/home/nikola/Git/rtl8723au_bt/rtk_btusb.c:41:2: error: #error "This repo is only for kernels older than 4.1.0. For kernel 4.1 or later, use the kernel branch."
 #error "This repo is only for kernels older than 4.1.0. For kernel 4.1 or later, use the kernel bra
  ^
/home/nikola/Git/rtl8723au_bt/rtk_btusb.c: In function ‘btusb_intr_complete’:
/home/nikola/Git/rtl8723au_bt/rtk_btusb.c:168:7: error: implicit declaration of function ‘hci_recv_fragment’ [-Werror=implicit-function-declaration]
   if (hci_recv_fragment(hdev, HCI_EVENT_PKT,
       ^
/home/nikola/Git/rtl8723au_bt/rtk_btusb.c: In function ‘btusb_close’:
/home/nikola/Git/rtl8723au_bt/rtk_btusb.c:575:18: error: ‘NUM_REASSEMBLY’ undeclared (first use in this function)
  for (i = 0; i < NUM_REASSEMBLY; i++) {
                  ^
/home/nikola/Git/rtl8723au_bt/rtk_btusb.c:575:18: note: each undeclared identifier is reported only once for each function it appears in
/home/nikola/Git/rtl8723au_bt/rtk_btusb.c:576:10: error: ‘struct hci_dev’ has no member named ‘reassembly’
   if(hdev->reassembly[i]) {
          ^
/home/nikola/Git/rtl8723au_bt/rtk_btusb.c:577:18: error: ‘struct hci_dev’ has no member named ‘reassembly’
    kfree_skb(hdev->reassembly[i]);
                  ^
/home/nikola/Git/rtl8723au_bt/rtk_btusb.c:578:8: error: ‘struct hci_dev’ has no member named ‘reassembly’
    hdev->reassembly[i] = NULL;
        ^
cc1: some warnings being treated as errors
scripts/Makefile.build:264: recipe for target '/home/nikola/Git/rtl8723au_bt/rtk_btusb.o' failed
make[2]: *** [/home/nikola/Git/rtl8723au_bt/rtk_btusb.o] Error 1
Makefile:1388: recipe for target '_module_/home/nikola/Git/rtl8723au_bt' failed
make[1]: *** [_module_/home/nikola/Git/rtl8723au_bt] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.4.6-301.fc23.x86_64'
Makefile:15: recipe for target 'all' failed
make: *** [all] Error 2
```

gssci, 2016-04-15:
Yep, same problem here

lwfinger, 2016-04-15:
There was a change in depmod that leads to this error. My distro (openSUSE) has not yet implemented this new version, thus I could not see the error until I created a Fedora 23 system.

The fix was trivial and has been pushed to GitHub. Do a new pull.

Sorry about the confusion about what branch to use. As the error statement says, you do need to use the kernel branch. I am currently having an interchange with another F23 user that appears to throw the kitchen sink at every problem and I got your thread mixed in my head with his.

nikolaandj, 2016-04-15:
Compilation and installation now works. But I am facing another problem:

When I boot my notebook, bluetooth is disabled, and cannot be enabled. Entering

```
sudo modprobe -rv btusb
sudo modprobe -v btusb
```

allows me to enable the bt, but I don't see any devices. On my phone, on the other hand, I can see the notebook, but I cannot connect. Here is the output of `dmesg | grep hci`

```
[    3.289062] ahci 0000:00:11.0: version 3.0
[    3.289426] ahci 0000:00:11.0: AHCI 0001.0300 32 slots 2 ports 6 Gbps 0x3 impl SATA mode
[    3.289431] ahci 0000:00:11.0: flags: 64bit ncq sntf ilck pm led clo pmp pio slum part 
[    3.290176] scsi host0: ahci
[    3.290411] scsi host1: ahci
[    3.290769] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    3.290776] ehci-pci: EHCI PCI platform driver
[    3.291411] ehci-pci 0000:00:12.2: EHCI Host Controller
[    3.291663] ehci-pci 0000:00:12.2: new USB bus registered, assigned bus number 1
[    3.291674] ehci-pci 0000:00:12.2: applying AMD SB700/SB800/Hudson-2/3 EHCI dummy qh workaround
[    3.291689] ehci-pci 0000:00:12.2: debug port 1
[    3.291757] ehci-pci 0000:00:12.2: irq 17, io mem 0xd684d000
[    3.297933] ehci-pci 0000:00:12.2: USB 2.0 started, EHCI 1.00
[    3.298034] usb usb1: Manufacturer: Linux 4.4.6-301.fc23.x86_64 ehci_hcd
[    3.299046] ehci-pci 0000:00:13.2: EHCI Host Controller
[    3.299158] ehci-pci 0000:00:13.2: new USB bus registered, assigned bus number 2
[    3.299163] ehci-pci 0000:00:13.2: applying AMD SB700/SB800/Hudson-2/3 EHCI dummy qh workaround
[    3.299174] ehci-pci 0000:00:13.2: debug port 1
[    3.299216] ehci-pci 0000:00:13.2: irq 17, io mem 0xd684b000
[    3.304706] ehci-pci 0000:00:13.2: USB 2.0 started, EHCI 1.00
[    3.304782] usb usb2: Manufacturer: Linux 4.4.6-301.fc23.x86_64 ehci_hcd
[    3.305394] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
[    3.305403] ohci-pci: OHCI PCI platform driver
[    3.305706] ohci-pci 0000:00:12.0: OHCI PCI host controller
[    3.305814] ohci-pci 0000:00:12.0: new USB bus registered, assigned bus number 3
[    3.305858] ohci-pci 0000:00:12.0: irq 18, io mem 0xd684e000
[    3.360832] usb usb3: Manufacturer: Linux 4.4.6-301.fc23.x86_64 ohci_hcd
[    3.361789] ohci-pci 0000:00:13.0: OHCI PCI host controller
[    3.361915] ohci-pci 0000:00:13.0: new USB bus registered, assigned bus number 4
[    3.361945] ohci-pci 0000:00:13.0: irq 18, io mem 0xd684c000
[    3.416829] usb usb4: Manufacturer: Linux 4.4.6-301.fc23.x86_64 ohci_hcd
[    3.417657] ohci-pci 0000:00:14.5: OHCI PCI host controller
[    3.417764] ohci-pci 0000:00:14.5: new USB bus registered, assigned bus number 5
[    3.417793] ohci-pci 0000:00:14.5: irq 18, io mem 0xd684a000
[    3.472826] usb usb5: Manufacturer: Linux 4.4.6-301.fc23.x86_64 ohci_hcd
[    3.473435] uhci_hcd: USB Universal Host Controller Interface driver
[    3.473717] xhci_hcd 0000:00:10.0: xHCI Host Controller
[    3.473797] xhci_hcd 0000:00:10.0: new USB bus registered, assigned bus number 6
[    3.474010] xhci_hcd 0000:00:10.0: hcc params 0x014040c3 hci version 0x100 quirks 0x00000418
[    3.474410] usb usb6: Manufacturer: Linux 4.4.6-301.fc23.x86_64 xhci-hcd
[    3.474822] xhci_hcd 0000:00:10.0: xHCI Host Controller
[    3.474881] xhci_hcd 0000:00:10.0: new USB bus registered, assigned bus number 7
[    3.477428] usb usb7: Manufacturer: Linux 4.4.6-301.fc23.x86_64 xhci-hcd
[    3.778699] usb 3-1: new low-speed USB device number 2 using ohci-pci
[    3.796558] usb 5-2: new full-speed USB device number 2 using ohci-pci
[    3.903690] usb 1-5: new high-speed USB device number 4 using ehci-pci
[    4.221517] usb 3-3: new full-speed USB device number 3 using ohci-pci
[   12.046680] Bluetooth: hci0: rtl: examining hci_ver=06 hci_rev=000b lmp_ver=06 lmp_subver=8723
[   12.046719] Bluetooth: hci0: rtl: loading rtl_bt/rtl8723b_fw.bin
[   12.049725] Bluetooth: hci0: rom_version status=0 version=1
[   27.825197] usb 3-3: reset full-speed USB device number 3 using ohci-pci
[  141.249502] Bluetooth: hci0: rtl: examining hci_ver=06 hci_rev=000b lmp_ver=06 lmp_subver=8723
[  141.249516] Bluetooth: hci0: rtl: loading rtl_bt/rtl8723b_fw.bin
[  141.252452] Bluetooth: hci0: rom_version status=0 version=1
[  155.133137] usb 3-3: reset full-speed USB device number 3 using ohci-pci
[  156.962997] Bluetooth: hci0 command 0x2005 tx timeout
[  158.966552] Bluetooth: hci0 command 0x200b tx timeout
[  160.970416] Bluetooth: hci0 command 0x200c tx timeout
```

Note the last three lines. They appear every time I try to scan for devices.

lwfinger, 2016-04-15:
Using this driver on my openSUSE system, I can scan for BT devices and pair with my cell phone, which is the only BT device I have. On the F23 virtual machine, I could scan for it using hcitool. The F23 BlueTooth GUI said I did not have any BT devices. I have no idea what that means, but as far as I can tell, the driver is working fine. I also have an Intel BT device, and the same thing happened on F23. I have no suggestions on how to debug this issue.

nikolaandj, 2016-04-15:
Are there any parameters as with the WiFi module? I used blueman in my 
tests. Hcitool reports: Inquiry failed: Connection timed out

On my phone I get the message: "Operation timed out" while trying to 
pair

lwfinger, 2016-04-15:
Do you mean WIFI or BT module. If the latter, I use the "reset=0" option.

To see the parameters for any module, use the modinfo command.

nikolaandj, 2016-04-15:
I will try that option as soon as I am available to. I will report my 
results here. Thanks for the info :)

leon9923, 2016-09-09:
+nikolaandj Did you find a fix for this this?

nikolaandj, 2016-09-09:
+leon9923 To be honest, I moved to openSUSE Leap 42.1 soon after I replied, and I didn't get a chance to try the parameter. I'll try it soon, since I am currently unable. **NOTE:** _I am currently on kernel 4.1.27._

nikolaandj, 2016-09-09:
+leon9923 Update: I have just enables my bluetooth in bios (since I have disabled it since my last reply), and on oS 42.1 I works out of the box. I haven't tried compiling the version from this repository. Which distro are you using, and which kernel version?

leon9923, 2016-09-09:
+nikolaandj  I'm on 4.4.0-36-generic
and linux mint 18 sarah
Looks Like I should downgrade

leon9923, 2016-09-09:
+nikolaandj looks Like I should just move to 4.1 kernel

nikolaandj, 2016-09-09:
Have you tried the default kernel drivers? What error do you get?

leon9923, 2016-09-09:
+nikolaandj default drivers work at first and then they stop completely

nikolaandj, 2016-09-09:
I haven't tested my drivers extensively, though. Also haven't tried
building the newer version. If I have new updates I will contact you.

Best regards :)

leon9923, 2016-09-09:
+nikolaandj thanks the most weird thing is there are no errors nothing that can give me any clue on what to do everything is fine but I still can't search nor any device can find the laptop I think some package is interfering with the drivers