lwfinger/rtl8723au_bt #52: rtl8723be (0bda:b001) bluetooth doesnt work on linux-4.6.0-1-amd64

xundeenergie, 2016-06-21:
Hi There!

I have this combinated WIFI/Bluetooth-Adapter on my HP ProBook 450 G2. And it worked for a long time, since Kernel 4.0 (i think i remember correcktly)

Now it stopped working on Debian stretch with kernel 4.6.0-1, also with Kernel 4.5

```
# lsusb
Bus 001 Device 002: ID 8087:8000 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 004: ID 062a:4101 Creative Labs Wireless Keyboard/Mouse
Bus 002 Device 003: ID 05c8:0374 Cheng Uei Precision Industry Co., Ltd (Foxlink) 
Bus 002 Device 002: ID 0bda:b001 Realtek Semiconductor Corp. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

I compiled and installed the rtlwifi from the troy-branch, and i can see my laptop on my mobile-phone, but i'm not able to pair the devices. The laptop does not recognize the mobile-phone on bluetooth.

Do you need more infos

Best regards

Jakob

xundeenergie, 2016-06-21:
I tried rtlwifi from kernel-branch on
# uname -r
4.6.0-1-amd64

after compiling and make install i removed the module with

modprobe -rv btusb

Then i reloaded the module

```
# modprobe -v btusb
insmod /lib/modules/4.6.0-1-amd64/kernel/drivers/bluetooth/btintel.ko 
insmod /lib/modules/4.6.0-1-amd64/kernel/drivers/bluetooth/btbcm.ko 
insmod /lib/modules/4.6.0-1-amd64/kernel/drivers/bluetooth/btrtl.ko 
insmod /lib/modules/4.6.0-1-amd64/kernel/drivers/bluetooth/btusb.ko 

```
journalctl -f gives me:
```
Jun 21 22:46:43 aldebaran kernel: usbcore: registered new interface driver btusb
Jun 21 22:46:43 aldebaran systemd[1]: Starting Load/Save RF Kill Switch Status...
Jun 21 22:46:43 aldebaran kernel: Bluetooth: hci0: rtl: examining hci_ver=06 hci_rev=000b lmp_ver=06 lmp_subver=8723
Jun 21 22:46:43 aldebaran kernel: Bluetooth: hci0: rtl: loading rtl_bt/rtl8723b_fw.bin
Jun 21 22:46:43 aldebaran kernel: bluetooth hci0: firmware: direct-loading firmware rtl_bt/rtl8723b_fw.bin
Jun 21 22:46:43 aldebaran kernel: Bluetooth: hci0: rom_version status=0 version=1
Jun 21 22:46:44 aldebaran bluetoothd[1183]: Failed to obtain handles for "Service Changed" characteristic
Jun 21 22:46:44 aldebaran systemd-udevd[10858]: failed to execute '/usr/bin/hciconfig' '/usr/bin/hciconfig hci0 up': No such file or directory
Jun 21 22:46:44 aldebaran systemd-udevd[10853]: Process '/usr/bin/hciconfig hci0 up' failed with exit code 2.
Jun 21 22:46:44 aldebaran bluetoothd[1183]: Not enough free handles to register service
Jun 21 22:46:44 aldebaran bluetoothd[1183]: Error adding Link Loss service
Jun 21 22:46:44 aldebaran bluetoothd[1183]: Not enough free handles to register service
Jun 21 22:46:44 aldebaran bluetoothd[1183]: Not enough free handles to register service
Jun 21 22:46:44 aldebaran bluetoothd[1183]: Not enough free handles to register service
Jun 21 22:46:44 aldebaran bluetoothd[1183]: Current Time Service could not be registered
Jun 21 22:46:44 aldebaran bluetoothd[1183]: gatt-time-server: Input/output error (5)
Jun 21 22:46:44 aldebaran bluetoothd[1183]: Not enough free handles to register service
Jun 21 22:46:44 aldebaran bluetoothd[1183]: Not enough free handles to register service
Jun 21 22:46:44 aldebaran bluetoothd[1183]: Sap driver initialization failed.
Jun 21 22:46:44 aldebaran bluetoothd[1183]: sap-server: Operation not permitted (1)
Jun 21 22:46:44 aldebaran systemd[1]: Reached target Bluetooth.
Jun 21 22:46:44 aldebaran bluetoothd[1183]: RFCOMM server failed for Headset Voice gateway: rfcomm_bind: Address already in use (98)
Jun 21 22:46:44 aldebaran systemd[1]: Started Load/Save RF Kill Switch Status.
Jun 21 22:46:44 aldebaran bluetoothd[1183]: Endpoint registered: sender=:1.237 path=/MediaEndpoint/A2DPSource
Jun 21 22:46:44 aldebaran bluetoothd[1183]: Endpoint registered: sender=:1.237 path=/MediaEndpoint/A2DPSink
Jun 21 22:46:44 aldebaran bluetoothd[1183]: Endpoint registered: sender=:1.67 path=/MediaEndpoint/A2DPSource
Jun 21 22:46:44 aldebaran bluetoothd[1183]: Endpoint registered: sender=:1.67 path=/MediaEndpoint/A2DPSink

```

You need more information?

rafa400, 2016-06-28:
Similar for me: Same laptop and working if my boot kernel is 4.5.0-2-amd64 and NOT working with 4.6.0-1-amd64