lwfinger/rtl8723au_bt #53: rtk8723au bt driver not working 4.4.0-36-generic

leon9923, 2016-09-07:
generic details: 
                        kernel-4.4.0-36-generic
                        Linux mint 18 sarah.
                          wifi driver rtl8723be

                        Things attached: Dmesg log, journalctl log, systemctl log.
Things that get the bluetooth device started are modprobe -rv btusb, modprobe -v btusb.
Module installed from kernel branch rest all throw errors.

Problem: I have installed rtlwifi .
 My phone scans the laptop after updating bluez to the latest version, unable to pair/connect and hcitool scan fails. 
After modprobe commands the bluetooth is visible for  a rough 30 secs.
I've read all the issues on your kernel thread alspo still couldn't find a way to keep this stable
could you guide me on how to fix this.
I know you're very busy but a little help would be sufficient for me.
  
dmesg log and journalctl log
[image].
.
.
.
.
.
.
systemctl log
[image]

bajbijendra, 2016-09-08:
thanks for sharing . this is solved then.screens shot are txt mode than better as easier to copy paste and search. here is another rtl driver for https://github.com/vincent-t/rt8192cu_dkms device -http://plugable.com/drivers/usb-wifint/

leon9923, 2016-09-08:
@bajbijendra  Thanks will try this out and let you know

leon9923, 2016-09-08:
@bajbijendra  My wifi driver works fine Bluetooth on the other hand doesn't these above drivers are for wifi only

leon9923, 2016-09-14:
@lwfinger, sorry If I'm causing any trouble, I was on 4.4 generic kernel I moved to 4.5 stable and I get the error in systemctl status bluetooth :: 
systemd[1]: Starting Bluetooth service...
bluetoothd[11068]: Bluetooth daemon 5.37
 bluetoothd[11068]: Starting SDP server
bluetoothd[11068]: Bluetooth management interface 1.1
 bluetoothd[11068]: Failed to obtain handles for "Serv
 bluetoothd[11068]: Failed to clear UUIDs: Busy (0x0a)
 systemd[1]: Started Bluetooth service.
 bluetoothd[11068]: Loading LTKs timed out for hci0
 bluetoothd[11068]: Failed to set mode: Invalid Index 
 bluetoothd[11068]: Failed to obtain handles for "Serv

*After building the rtl8723au kernel branch*

Sep 15 01:25:23 afro-coder systemd[1]: Starting Bluetooth service...
Sep 15 01:25:23 afro-coder bluetoothd[18635]: Bluetooth daemon 5.41
Sep 15 01:25:23 afro-coder systemd[1]: Started Bluetooth service.
Sep 15 01:25:23 afro-coder bluetoothd[18635]: Starting SDP server
Sep 15 01:25:23 afro-coder bluetoothd[18635]: Bluetooth management interface 1.1
Sep 15 01:25:23 afro-coder bluetoothd[18635]: Failed to obtain handles for "Serv
 ESCOC

*If I move back to bluez 5.23*

Sep 15 01:29:23 afro-coder systemd[1]: Starting Bluetooth service...
Sep 15 01:29:23 afro-coder bluetoothd[23188]: Bluetooth daemon 5.23
Sep 15 01:29:23 afro-coder bluetoothd[23188]: Starting SDP server
Sep 15 01:29:23 afro-coder systemd[1]: Started Bluetooth service.
Sep 15 01:29:23 afro-coder bluetoothd[23188]: Bluetooth management interface 1.1

Still no connections nor can any device detect

develCuy, 2018-08-24:
it works for me 4.4.0-131-generic