lwfinger/rtl8723au_bt #54: branch:kernel CONFIG_BT_RTL vs CONFIG_BT_HCIBTUSB_RTL

ferbar, 2016-10-27:
I think this should be the same flag. Without CONFIG_BT_HCIBTUSB_RTL the firmware is never loaded for devices with .driver_info = BTUSB_REALTEK

lwfinger, 2016-10-27:
In the kernel, that extra flag is defined. I have modified the code here to only use CONFIG_BT_RTL.

closed 2016-10-27 by lwfinger

ferbar, 2016-10-27:
This is what I have to change to get the bluetooth part of rtl8723bu working:

patch btusb.c (8723BU and many more is missing in btusb.c/kernel branch) and enable
  hdev->setup = btrtl_setup_realtek; below

```
diff --git a/btusb.c b/btusb.c
index 870dda7..1a90d99 100644
--- a/btusb.c
+++ b/btusb.c
@@ -329,6 +329,12 @@ static const struct usb_device_id blacklist_table[] = {
        /* Silicon Wave based devices */
        { USB_DEVICE(0x0c10, 0x0000), .driver_info = BTUSB_SWAVE },
 
+       // [chris]
+
+       /*8723BU*/
+       { USB_DEVICE(0x0bda, 0xb720), .driver_info = BTUSB_REALTEK },
+       { USB_DEVICE(0x0bda, 0xb72a), .driver_info = BTUSB_REALTEK },
+
        { }     /* Terminating entry */
 };
 
@@ -2741,7 +2747,7 @@ static int btusb_probe(struct usb_interface *intf,
                hdev->set_bdaddr = btusb_set_bdaddr_ath3012;
        }
 
-#ifdef CONFIG_BT_HCIBTUSB_RTL
+#ifdef CONFIG_BT_RTL
        if (id->driver_info & BTUSB_REALTEK) {
                hdev->setup = btrtl_setup_realtek;
 

```

make, make install, start the wpa_supplicant, hciconfig up, hciconfig piscan

now: hcitool scan works and the device is found from my phone.

without the patch nothing is found

ferbar, 2016-10-27:
ok.... CONFIG_BT_HCIBTUSB_RTL exists since 4.2, see: http://cateee.net/lkddb/web-lkddb/BT_HCIBTUSB_RTL.html