lwfinger/rtl8723au_bt #67: Can't get this to work. No matter what

linuxy9, 2018-04-25:
I have updated my kernal and a new install of Linux Mint to once was a intel atom skylake tablet. Everything was okay until i need a mouse and keyboard witch i havr bu the bluetooth does not work. I must tell you I am a noob at this most definitely. 

What ive tried doing

+ Compled the Drivers
+ Wifi works
-  bluetooth servive does not start even with `hciattach`

This is a reltek bluetooth and wifi device

Cant remember the name of it rn since my tablet is wall charging.

I desperately need help and this newest kernal.

lwfinger, 2018-04-25:
What kernel version are you running? With something as new as 4.15+, the BT driver is there and it will work IF the firmware is available.  Look in the dmesg output for clues.

linuxy9, 2018-04-26:
Okay

linuxy9, 2018-04-26:
seems i have posted this in the wrong repository. 

error i get:
```
linuxy-DPW8A-BT rtl8723bs_bt # /usr/sbin/rtk_hciattach -n -s 115200 ttyS5 rtk_h4 
Realtek Bluetooth init uart with init speed:115200, final_speed:115200, type:HCI UART H4
Realtek Bluetooth :Realtek hciattach version 2.5 

Realtek Bluetooth :config offset(f4),length(8)
Realtek Bluetooth :config baud rate to :4928002, hwflowcontrol:5f, 1
Realtek Bluetooth :config offset(27),length(1)
Realtek Bluetooth :config offset(fe),length(1)
Realtek Bluetooth :config offset(15b),length(4)
Realtek Bluetooth :config offset(1e3),length(1)
Realtek Bluetooth :Get config baud rate from config file:4928002
Realtek Bluetooth :Load FW OK
Realtek Bluetooth ERROR: Check signature error!
Realtek Bluetooth :baudrate in change speed command: 0x2 0x80 0x92 0x4 

Realtek Bluetooth :H4 Change uart Baudrate after write 
Initialization timed out.
```

linuxy9, 2018-04-26:
output of ` uname -a`

```
linuxy-DPW8A-BT rtl8723bs_bt # uname -a
Linux linuxy-DPW8A-BT 4.15.0-15-generic #16~16.04.1-Ubuntu SMP Thu Apr 5 12:19:23 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```

please help if you can.