lwfinger/rtl8723au_bt #74: Does the driver support Linux Kernel 5.4 (LTS) ?

mochnaufals, 2021-04-11:
Hello,

I've been using this driver for about 1 year in my Linux Mint. It worked for me. But then, I was switching to windows for another year and now I am dualbooting my Windows with Manjaro Linux. I followed the instructions in https://github.com/lwfinger/rtl8723au_bt/issues/40 (when I was using Linux Mint, this 'hack' worked for me). But now, I got an error saying:
`No rule to make target 'bluetooth'`
I installed the proper linux kernel headers for my kernel (5.4 LTS) but I stil cannot install the driver (still got the same error). I also did some research regarding this Bluetooth Driver issue (my laptop is Toshiba Satellite C55D), but what I get are older posts (more than 3 years ago).

Hence, I wonder if this driver still works for kernel 5.4 LTS.
Any suggestions?

Many thanks,
Naufal

lwfinger, 2021-04-11:
For kernels as new as 5.4, the btusb and rtlusb drivers built into the kernel should work without needing an external driver. Is this not the case?

The error message you are quoting looks like a makefile error. As this driver is not maintained, that is certainly a possibility.

mochnaufals, 2021-04-11:
> For kernels as new as 5.4, the btusb and rtlusb drivers built into the kernel should work without needing an external driver. Is this not the case?
> 
> The error message you are quoting looks like a makefile error. As this driver is not maintained, that is certainly a possibility.

Unfortunately, I cannot use my Bluetooth after fresh install my Manjaro Linux. I used the 5.10 and 5.4 kernels but alao didn't work. I also switched off my Bluetooth in my Windows 10 OS but still didn't work

lwfinger, 2021-04-12:
I just checked my rtl8723au_bt repo, and it compiled correctly. That error is likely being caused by something in your distro.

Please do a 'make clean' and 'make'. Send me everything that shows on your terminal.

mochnaufals, 2021-04-12:
Here is the output,

`[mochnaufals@manjarosatellite rtl8723au_bt-master]$ sudo make clean
rm -rf *.o *.mod.c *.mod.o *.ko *.symvers *.order *.a
[mochnaufals@manjarosatellite rtl8723au_bt-master]$ sudo make
make -C /lib/modules/5.4.108-1-MANJARO/build M=/home/mochnaufals/Downloads/driver bluetooth toshiba/rtl8723au_bt-master modules
make[1]: Entering directory '/usr/lib/modules/5.4.108-1-MANJARO/build'
make[1]: *** No rule to make target 'bluetooth'.  Stop.
make[1]: Leaving directory '/usr/lib/modules/5.4.108-1-MANJARO/build'
make: *** [Makefile:15: all] Error 2
`

mochnaufals, 2021-04-12:
Here is the output for `sudo make clean`,
`rm -rf *.o *.mod.c *.mod.o *.ko *.symvers *.order *.a`

And here is the output for `sudo make`
`make -C /lib/modules/5.4.108-1-MANJARO/build M=/home/mochnaufals/Downloads/driver bluetooth toshiba/rtl8723au_bt-master modules
make[1]: Entering directory '/usr/lib/modules/5.4.108-1-MANJARO/build'
make[1]: *** No rule to make target 'bluetooth'.  Stop.
make[1]: Leaving directory '/usr/lib/modules/5.4.108-1-MANJARO/build'
make: *** [Makefile:15: all] Error 2`

lwfinger, 2021-04-12:
The make line should be
make -C /lib/modules/5.11.0-rc6-02523-g9e2fd29864c5-dirty/build M=/home/finger/rtl8723au_bt modules

That bluetooth there is wrong!! Have you changed the Makefile?

mochnaufals, 2021-04-12:
I did not change anything. I also downloaded the file again and try make, still got the same error. I see that my working directory has space, probably that's the problem.

Let me try again and I will inform you.
Thank you in advance

mochnaufals, 2021-04-12:
@lwfinger Hello, I renamed my working directory (folders) so the Makefile can be executed. After that, the script can be executed and seems running properly. Before I ran it, I follow the instructions from https://github.com/lwfinger/rtl8723au_bt/issues/40 (add one line of my vendor ID). The script seems to be fine. I also follow the instruction guide from Readme to do sudo modprobe. But then, the bluetooth still scanning without finding a device (I tried several different devices, still not works). I rebooted my laptop, and execute sudo modprobe again, but still not working.

Do you have any suggestions?

closed 2021-04-12 by lwfinger